NavAbility/NavAbilitySDK.jl #42: Release v0.1.0

GearsAD, 2022-01-17:
@JuliaRegistrator register(branch="release/v0.1")

GearsAD, 2022-01-17:
@JuliaRegistrator register(branch="release/v0.1")

JuliaRegistrator, 2022-01-17:
Registration pull request created: [JuliaRegistries/General/52614](https://github.com/JuliaRegistries/General/pull/52614)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.0 -m "<description of version>" c8cb84fcf7c95d72faaf7fb34ecefbd090195551
git push origin v0.1.0
```

JuliaRegistrator, 2022-01-17:
Registration pull request updated: [JuliaRegistries/General/52614](https://github.com/JuliaRegistries/General/pull/52614)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.0 -m "<description of version>" e7ec1b5406d157c6f76e43b7620a054cd14df9b2
git push origin v0.1.0
```

closed 2022-01-20 by jim-hill-r