NavAbility/NavAbilitySDK.jl #74: To simplify understanding remove the `Data` suffix from all the FactorData generator functions

GearsAD, 2022-03-26:
Hard pressed to see the value in the suffix.

jim-hill-r, 2022-03-27:
You have to designate the difference between a Factor and it's data cuz one is fed into the other. I think it's really confusing to do:

`
x1x2f1InputStuff = Pose2Factor(....)
x1x2f1 = Factor('x1x2f1', x1x2f1InputStuff)
addFactor(x1x2f1)
`

This is much clearer

`
X1x2f1Data = Pose2FactorData(...)
x1x2f1 = Factor('x1x2f1', x1x2f1Data)
addFactor(x1x2f1)

`

dehann, 2022-03-27:
Also the cousin over by `VariableNodeData` inside `DFGVariable / Variable`

Affie, 2022-04-14:
We currently have:
```julia
f = Factor("x0x1f1", "Pose2Pose2", ["x0","x1"], 
  Pose2Pose2Data(
    Z=FullNormal(
      [1.0, 0.0, pi/2], 
      diagm([0.1, 0.1, 0.01].^2))
  )
)
addFactor(client, context, f)
```

Is something like this not possible as a wrapper?:

```julia 
addFactor(client_context, ["x0","x1"],  Pose2Pose2(FullNormal(...)); kwargs...)
```

dehann, 2022-04-15:
Yeah, that's what I want it to look like.  Please do that.

I would also name the first argument `platform` instead of `client_context`.  That way in the future when users are talking with an Edge sidecar, its still just "platform" and a config file that specifies where solves should be happening.

Affie, 2022-04-15:
yes, `client_context` was just to show its one to combine the 2. 
Maybe it should be a separate issue, are we happy with `NavAbilityPlatform` as the name from https://github.com/NavAbility/NavAbilitySDK.jl/pull/78
The internals and constructors can still change, but if we can decide on the name it would be a step in the right direction.

jim-hill-r, 2022-04-15:
If we are changing signatures can we please have a design discussion. This is going away from existing plans.

Affie, 2022-04-16:
The plan is not to change signatures, but to add a wrapper that resembles IIF.  The signatures in IIF has had a lot of iterations and has become rather  stable. So I think its a good starting point for the wrappers. 

> This is going away from existing plans.

Can you please point me to the existing plans to get an better understanding on it.

closed 2022-05-11 by dehann