NavAbility/NavAbilitySDK.jl #77: SDKs should not create factor labels

dehann, 2022-03-27:
SDKs should not have to define factor labels at the edge, e.g. “x0l1f1” – this is automatically done by IIF.addFactor.

Is there a specific reason?

jim-hill-r, 2022-03-27:
I believe it is required in the packed data. Also currently I think we use update factor to actually do the add. This will be solved elegantly when we implement fully typed GraphQL API. Right now it just packed data.

Also on a personal level, I know that appending f# is standard practice but I have a really hard time with the idea of being forced to do naming the way the framework wants me to. I am a verbose namer and not having choice to call my factor StarttoAprilLandmark1 or something meaningful drives me a bit bonkers. It's a long term concern and has other implications. I also think requiring labels to be unique is a bit silly. Just me. Most databases will store an ID and a display name. We are using label for both.

dehann, 2022-03-27:
Shall we chew on the factor naming ideas and thoughts in a discussion here:
- https://github.com/JuliaRobotics/Caesar.jl/discussions/826