NavAbility/NavAbilitySDK.jl #79: Tests for 1D and Mixture cases.

GearsAD, 2022-03-27:
There are not integration tests for these cases and there should be some before we tag this release.

GearsAD, 2022-03-29:
REF: https://github.com/NavAbility/NavAbilitySDK.py/blob/main/tests/conftest.py#L82-L140

dehann, 2022-03-29:
I'm going to update the commented 1D graph sectionin in test/integration/fixtures.jl to clear this.

What I'm going to recommend medium term, is that we build one test harness repo for the SDKs, using the pre-existing standardized API as best we can, and then use that as an upstream repo for CI on all the SDKs.  That way, if we change something in the test, then all the SDKs will see the change and we can fix accordingly.

PR #87 (2022-03-30): Test 1D and Mixture,
closes #79
commit 66e0fd9f (2022-03-30): Test 1D and Mixture,

closes #79
closed 2022-03-30 by dehann