NavAbility/NavAbilitySDK.jl #81: Refactor integration tests into scenarios (e.g. 1D, 2D, etc.) so we can leverage representative cases for testing.

GearsAD, 2022-03-27:
As a developer I want access to a 1D fixture that gives me everything I need to perform a simple functional integrated test.

GearsAD, 2022-03-28:
We need to test:
- Point2 and BearingRange for Tutorial 1 
- ContinuousScalar, LinearRelative, and the various distributions for Tutorial 2
- Bearing for Tutorial 3

dehann, 2022-03-29:
I'm planning to develop the tests using this code as starting point:

https://github.com/NavAbility/BinderNotebooks/issues/43#issue-1182397606

jim-hill-r, 2022-03-29:
My goal originally was to have demos be our regression. With the pivot to Notebooks, we would have to figure out how to run and add assertions to Notebooks. Likely not before ICRA, just a note on the strategy I was attempting to implement.

jim-hill-r, 2022-03-30:
I believe this is being worked by Dehann but it is not in progress. Please let me know if I can work it.

dehann, 2022-03-31:
Oh i did not get to this and only noticed @jim-hill-r s post now, sorry about that.  Yes please Jim, if you could work this please.  I'm going to reassign to you.

I took it on the weekend to help, ended up building the 1D and mixture fixture test.  That was the bigger blocker, and actually not 100% up to speed with ideas you might have add for this refactor integration title.  So best if you take it please.

Separately, as part of future API standardization etc.  I think we should make an upstream test harness repo for the SDKs and then have them all test against that.  I'd like to have an Engineering meeting on that first though, just mentioning know to socialize my thoughts for the future work.

jim-hill-r, 2022-03-31:
> Oh i did not get to this and only noticed @jim-hill-r s post now, sorry about that. Yes please Jim, if you could work this please. I'm going to reassign to you.

K thanks. I'll work it.

> Separately, as part of future API standardization etc. I think we should make an upstream test harness repo for the SDKs and then have them all test against that. I'd like to have an Engineering meeting on that first though, just mentioning to socialize my thoughts well in advance :-)

Yes let's do a testing session in engineering soon and document practices in coding-standards.

dehann, 2022-04-03:
@jim-hill-r , is this needed for Tut1 MVP?  Tut1 is not 1D, I take this Issue as technical diligence to build good testing.  Looks like you unassigned to rather work this issue as part of Eng discussion on a common upstream testing harness repo for the SDKs?

jim-hill-r, 2022-04-03:
I unassigned because I was not working it in favor of other more pressing issues. Feel free to move it or take it as needed.

dehann, 2022-04-04:
xref:
- https://github.com/JuliaRobotics/Caesar.jl/pull/831