NavAbility/NavAbilitySDK.jl #86: Julia SDK 0.4 release

GearsAD, 2022-03-29:


dehann, 2022-04-03:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2022-04-03:
Error while trying to register: **Register Failed**
@dehann, it looks like you are not a publicly listed member/owner in the parent organization (NavAbility).
If you are a member/owner, you will need to change your membership to public. See [GitHub Help](https://help.github.com/en/articles/publicizing-or-hiding-organization-membership)

JuliaRegistrator, 2022-04-03:
Registration pull request created: [JuliaRegistries/General/57841](https://github.com/JuliaRegistries/General/pull/57841)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.0 -m "<description of version>" c8a087dc40a7a6cdfd92cec651c4e71f98e0709c
git push origin v0.4.0
```

closed 2022-04-03 by dehann