NavAbility/NavAbilitySDK.jl #89: add function `isInitialized`

dehann, 2022-03-30:
Hi, could we add the function `isInitialized` to the SDK?
The CJL call is as follows:
```julia
isInitialized(fg, :x0)
```

jim-hill-r, 2022-03-30:
for tut 1? To what end?

dehann, 2022-03-30:
The graphinit process is described in Tutorial 1, however, I just moved it to Above and Beyond path since this is easier to do in the CJL version.  So SDKs versions of Tut1 will have to be a little different for the sake of MVP on SDK versions.

> To what end

This is a developer function often used.  A part of Tut1 is based on explaining how graphinit works.  I'm going to modify the SDK versions to instead just do a solve instead, and thereby get numerical values in the nodes rather than reckon on init values.

Will circle back in Above and Beyond list.

dehann, 2022-03-30:
to be clear, this function will have to be built at some point, it would have been good to have for Tut1, but can work around it for SDK versions easy enough, so does not qualify for MVP.  Hence why I moved it out.

Affie, 2023-04-19:
why not use solver data eg.
`getVariableSolverData(fgclient, :x0).initialized`

or we can add this to DFG:
```julia
function isInitialized(dfg::AbstractDFG, label::Symbol, key::Symbol=:default)
    return getVariableSolverData(dfg, label, key).initialized
end
```