NavAbility/NavAbilitySDK.jl #95: Wrong unit test for Mixture packing

Affie, 2022-04-05:
https://github.com/NavAbility/NavAbilitySDK.jl/blob/c8a087dc40a7a6cdfd92cec651c4e71f98e0709c/test/unit/testFactors.jl#L42-L47

dehann, 2022-04-07:
oops, could you clarify a bit more -- i'm not following the issue?

Affie, 2022-04-07:
I looked at the source code and this was not the part that I had wrong, I'ts only misleading. Will close.

Affie, 2022-04-07:
I looked at the unit test to figure out how to create a mixture factor, but it didn't work. Sam helped and we ended up with:
```julia
md = MixtureData("LinearRelative", (slip=Rayleigh(5), noslip=Uniform(35, 60)), [0.4, 0.6], 2)
mmo = Factor("x1x2f1", "Mixture", ["x1","x2"], md)
```

So I think the `LinearRelativeData` part should be `"LinearRelative"`

closed 2022-04-07 by Affie