NavAbility/NavAbilitySDK.jl #144: restore solveParametric test

dehann, 2022-08-17:


dehann, 2022-08-18:
@jim-hill-r , can you please take over the PR and ask @Affie for help as necessary to get this one test working please.

jim-hill-r, 2022-08-18:
@dehann I think root cause is the variable needs a "parametric" key added with the variable. Still debugging, but just a heads up. If I use IIF to pack the variable everything works great and the packed variable comes with a "parametric" solver key. If we do manual packing like in the SDK then issues arise.

```
cloud-system-srvc-iif-worker-1        | [ Info: Using parametric
cloud-system-srvc-iif-worker-1        | [ Info: x1 is not initialized, and will do so now...
cloud-system-srvc-iif-worker-1        | ┌ Warning: Factor x0x1f1 will be an orphan in the destination graph, and therefore not added.
cloud-system-srvc-iif-worker-1        | └ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/AMlMp/src/services/AbstractDFG.jl:990
cloud-system-srvc-iif-worker-1        | [ Info: x0 is not initialized, and will do so now...
cloud-system-srvc-iif-worker-1        | ┌ Error: Variable does not have solver data parametric
cloud-system-srvc-iif-worker-1        | └ @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/AMlMp/src/services/DFGVariable.jl:420
cloud-system-srvc-iif-worker-1        | [ Info: ErrorException("type Nothing has no field val")
cloud-system-srvc-iif-worker-1        | [ Info: ErrorException("type Nothing has no field val")
```

jim-hill-r, 2022-08-18:
@dehann I have fixed up both branches with the MRE's. I hope that is enough to close out this issue and get all tests passing end to end. If you have any problems, feel free to open up branches or issues and I can work on them.

closed 2022-08-26 by dehann