NavAbility/NavAbilitySDK.jl #265: Rename `addBlob!(..., filepath::AbstractString, ... )` to `uploadFile` and implement `addBlob!` that matches the api

Affie, 2024-10-23:
We should rather call this `uploadFile` and have an `addBlob!` that matches the api. 
I haven't had a use case for `uploadFile` yet, but it's good to have as I can see some uses.

_Originally posted by @Affie in https://github.com/NavAbility/NavAbilitySDK.jl/pull/264#discussion_r1812474147_

dehann, 2024-10-23:
Old addBlob! Is still there unchanged.  This is a different signature.  Signature parameter tells user it's a file.  End result is still that a blob was added.

Affie, 2025-03-18:
To work seamlessly across the different back-ends signatures such as `addBlob!` should look the same, ie. `addBlob!(store::AbstractBlobStore, blobId::UUID, blob)`