NavAbility/NavAbilitySDK.jl #297: Missing field or signature `description`

dehann, 2025-07-10:
```julia
sfg = DFG.buildSubgraph(dfg, ls(dfg), 2)

ERROR: type NavAbilityDFG has no field description
Stacktrace:
 [1] getproperty
   @ ./Base.jl:49 [inlined]
 [2] getDescription(dfg::NavAbilityDFG{VariableDFG, FactorDFG})
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:78
 [3] getDFGInfo(dfg::NavAbilityDFG{VariableDFG, FactorDFG})
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:55
 [4] deepcopyGraph(::Type{…}, sourceDFG::NavAbilityDFG{…}, variableLabels::Vector{…}, factorLabels::Vector{…}; graphLabel::Symbol, sessionId::Nothing, kwargs::@Kwargs{})
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:1174
 [5] buildSubgraph(::Type{…}, dfg::NavAbilityDFG{…}, variableFactorLabels::Vector{…}, distance::Int64; solvable::Int64, graphLabel::Symbol, sessionId::Nothing, kwargs::@Kwargs{})
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:1372
 [6] buildSubgraph(::Type{…}, dfg::NavAbilityDFG{…}, variableFactorLabels::Vector{…}, distance::Int64)
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:1354
 [7] buildSubgraph(dfg::NavAbilityDFG{…}, variableFactorLabels::Vector{…}, distance::Int64; kwargs::@Kwargs{})
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:1382
 [8] buildSubgraph(dfg::NavAbilityDFG{VariableDFG, FactorDFG}, variableFactorLabels::Vector{Symbol}, distance::Int64)
   @ DistributedFactorGraphs ~/.julia/packages/DistributedFactorGraphs/lIoPH/src/services/AbstractDFG.jl:1376
 [9] top-level scope
   @ REPL[12]:1
Some type information was truncated. Use `show(err)` to see complete types.
```

Affie, 2025-07-24:
`getDFGInfo` needs to be extended for a `NavAbilityDFG`. However, the new design splits the "DFGInfo" into an `Agent` and a `FactorgraphRoot` (name TBD). Therefore, this will change to using something like `getAgent` and `getGraphRoot`. That will get the containers with all the info that was previously in `getDFGInfo` as well as BlobEntries that is currently lost.

dehann, 2025-07-31:
Should this ticket move to "Info & Refine"?